paparazzi/paparazzi #2807: Bebop Orange Avoider (fdm gazebo) position flicker in GCS

matteobarbera, 2021-12-06:
While testing the `bebop_orange_avoider` and `bebop_orange_avoider_guided` in simulation I noticed that the drone position was not being displayed correctly in the GCS

[image] [image]

This bug seems to have been introduced during the autopilot reorg pull request #2721.

After some attempted debugging with @OpenUAS we found out a couple of possible sources for this issue. The position flicker is reflected in the ROTORCRAFT_FP message, and it seems to be caused by the `NPS_BYPASS_INS` variable, defined in file [conf/simulator/gazebo/airframes/bebop.xml](https://github.com/paparazzi/paparazzi/blob/master/conf/simulator/gazebo/airframes/bebop.xml).

[image]

While setting `NPS_BYPASS_INS` to false stops the flicker, there is then a mismatch between the position of the drone in the GCS and in Gazebo, which leads us to believe there is a conflicting value in the initialization.

We attempted to compare  these airframes to the `ardrone2_gazebo` airframe, which also uses
`<module name="fdm" type="gazebo"/>` but does not exhibit the same behavior. We saw that for the ardrone2 the `<section name="SIMULATOR">` block there is a dedicated file for the sensor parameters
https://github.com/paparazzi/paparazzi/blob/6e6e2f11ce7c18c8eabf5a2233fb80d378d4e102/conf/airframes/examples/ardrone2_gazebo.xml#L194
which hardcodes different variables related to sensor timings while for the bebop this is set to the magical 512Hz

https://github.com/paparazzi/paparazzi/blob/6e6e2f11ce7c18c8eabf5a2233fb80d378d4e102/conf/simulator/nps/nps_sensors_params_default.h#L34-L42

so possibly there are multiple issues with the orange_avoider airframes. We found a very similar sounding issue posted by @tomvand #2134, maybe the autopilot reorg reintroduced it?

gautierhattenberger, 2021-12-06:
Which INS are you using ? It seems to be ins_extended, but then it is really strange. This filter is only setting the position at a single place in the code and if the bypass is activated, it should be immediately overwritten by NPS.
Concerning the mismatch between Gazebo and the INS, I really don't know since I have never used Gazebo. Printing the reference points in each case is probably a good start.
In any case, the underlying have never been solved, and the reorg might have revealed it again.

gautierhattenberger, 2021-12-06:
with NPS/JSBSIM, I don't see any glitches with the bypass

matteobarbera, 2021-12-07:
The latest master was used, so yes, ins_extended. I specifically compared it to ardrone2_gazebo because it's one of the few airframes using fdm Gazebo as opposed to JSBSIM to try and find the source of the bug. I don't know if there is a specific reason to use fdm Gazebo, perhaps the easiest thing would be to convert the last few airframes to JSBSIM and only use Gazebo as a visualization tool

gautierhattenberger, 2021-12-07:
I don't use Gazebo, so I can't really evaluate to work to mix JSBSim and Gazebo for the video only. I think it is worth investigating the position difference with bypass or not.
The filtering of the source updating the state interface is a complex issue that should be addressed, but later

tomvand, 2021-12-10:
Hi @matteobarbera , can you share a branch or commit where this problem occurs? I have some time today to look at it (but no promises that I'll fix it, my holiday starts this weekend ;) )

matteobarbera, 2021-12-10:
@tomvand I just tested it and this occurs on the latest master 6e6e2f11ce7c18c8eabf5a2233fb80d378d4e102

tomvand, 2021-12-10:
Ok, I can replicate the issue here (commit 6e6e2f1 (master); tu delft course conf; bebop orangeavoid aircraft). The position in the GCS is incorrect, sometimes rapidly switching between two offsets; but otherwise the drone behavior is correct (including its behavior at the edge of the Gazebo cyberzoo). Disabling the INS bypass leads to unpredictable behavior on my pc.
I suspect a mismatch between the coordinate systems in paparazzi (probably the flight plan) and Gazebo...

tomvand, 2021-12-10:
What I've found so far is that there is an error in the lat/lon coordinates reported by the drone and those set in the flight plan and gazebo's spherical coordinates. Disabling the GPS (`<define name="NPS_NO_GPS" value="TRUE"/>`) stops the jumping, but I've not been able to unkill the drone yet despite adding `<define name="NO_GPS_NEEDED_FOR_NAV" value="TRUE"/>`. After disabling the simulated GPS, the INS message and NPS_SPEED_POS report the correct position in the ltp, but NPS_POS_LLH reports a slightly incorrect latitude.
Possibly there might be a mismatch when converting to/from spherical coordinates. The fdm uses a function supplied by Gazebo, which may produce different results than the one used further downstream in paparazzi.

tomvand, 2021-12-10:
I think I have found and (partially) fixed the problem, please test the proof-of-concept here: https://github.com/paparazzi/paparazzi/compare/master...tomvand:fix-2807?expand=1

The current nps_fdm_gazebo code uses Gazebo's spherical transformations to derive ECEF and LLA coordinates from Gazebo's XYZ coordinate frame (which is defined equal to paparazzi's LTP). Apparently these produce different results than paparazzi's implementation, causing inconsistencies between the different coordinate types reported by nps.

In the above branch I have replaced Gazebo's transform by paparazzi's **for the position**, and now the GCS position appears correctly for me. The same modifications would need to be applied to the velocity and other vectors before definitively merging this branch.

@matteobarbera can you please test that this solution works for you?

gautierhattenberger, 2021-12-10:
Note that JSBSim also have a spherical model (but can still output geodetic coordinates if needed), so we have a `gc_of_gd_lat_d` function in `math/pprz_geodetic_double.h` used during initialization. Not sure it helps here.

commit f9ca2ff4 (2022-01-18): Fix #2807 inconsistent coordinate transform in nps_fdm_gazebo (#2808)

* FIX #2807 by using pprz geodetic transformations instead of Gazebo's

* Cleanup

* Remove outdated section in nps_fdm_gazebo

* (WIP) clean up position transform and change to double accuracy

* Replace fdm_gazebo's spherical functions by paparazzi's
closed 2022-01-18 by gautierhattenberger