paparazzi/paparazzi #2865: ins_ekf2 is incompatible with fixedwing firmware

matteobarbera, 2022-04-20:
Currently including the module `ins_ekf2` will only compile for a `rotorcraft` airframe, due to a couple of conflicts:

#### Waypoints
I'm not 100% that `nav.h` is included for all fixedwing airframes, but when it is it will also include `nav_common.h`, which has a number of conflicting defines with `waypoints.h`

https://github.com/paparazzi/paparazzi/blob/e82db2530544ebc1a430c893fa45ff2791436412/sw/airborne/modules/nav/waypoints.h#L42-L47
https://github.com/paparazzi/paparazzi/blob/e82db2530544ebc1a430c893fa45ff2791436412/sw/airborne/modules/nav/waypoints.h#L51

https://github.com/paparazzi/paparazzi/blob/e82db2530544ebc1a430c893fa45ff2791436412/sw/airborne/modules/nav/common_nav.h#L39-L43
https://github.com/paparazzi/paparazzi/blob/e82db2530544ebc1a430c893fa45ff2791436412/sw/airborne/modules/nav/common_nav.h#L57

The definition of the variable `waypoints` as well as a number of macros that perform operations on waypoints are both double and use a different structure. I think it would make sense to move all waypoints and functions on waypoints to `waypoints.h` and include this file in `common_nav.h`.

#### Rotocraft guidance functions
Certain variables that are used in ekf2 seem to only be defined in rotorcraft stabilization and guidance files, `stab_att_sp`, `nav_heading`, `guidance_h_read_rc`, `stabilization_attitude_enter`

fvantienen, 2022-04-20:
The waypoints are only included for the `waypoints_localize_all()` function so not sure what the problem is there, but should be easily fixable.

The second part of rotorcraft guidance functions is there when the EKF2 decides to reset it's heading. You can disable it with `NO_RESET_UPDATE_SETPOINT_HEADING`, ideally we should implement something like an ABI call suich that the controllers can be notified if the heading gets reset.

gautierhattenberger, 2022-04-20:
The structure for the waypoints of fixedwing and rotorcraft is completely different and not not compatible (and of course, not easy to merge, that would be too simple). Even the "nav_common" is not common at all.
I think the problem is that a INS filter should not depend on navigation, although we can't always avoid that, for the init point for instance. If needed, you can use the defines `FIXEDWING_FIRMWARE`, `ROTORCRAFT_FIRMWARE` and `ROVER_FIRMWARE` to make conditional compilation.