paparazzi/paparazzi #2895: Installation issues Ubuntu 22 (libSDL.so)

tomvand, 2022-09-02:
Hi,

Due to laptop problems I have to reinstall paparazzi on a new system with Ubuntu 22, and here I'm running into some issues. I have manually installed git, build-essential and the likes, and then follow the one-liner instructions on the wiki here: https://wiki.paparazziuav.org/wiki/Installation (the Ubuntu 20 and above version). Unfortunately, running `make` seems to have two new issues that did not occur on my older Ubuntu 18 setup:

1. OCaml compilation gives a labels omitted warning: `Warning 6 [labels-omitted]: label callback was omitted in the application of this function.`
2. OCaml compilation of `input2ivy` fails because `libSDL.so` can't be found: 

```
File "_none_", line 1:
Error: I/O error: libSDL.so: No such file or directory
make[1]: *** [Makefile:78: input2ivy] Error 2
make: *** [Makefile:168: joystick] Error 2
```

Since this is a completely fresh install of Ubuntu 22, I suspect there is a dependency missing in the installation one-liner/documentation, or something has changed or moved in Ubuntu 22 or perhaps its ocaml packages.

I've uploaded the full ouput of the one-liner here: https://gist.github.com/tomvand/01b379e2ddc650575f814c66289f70f0 

libSDL.so is present on the system:
```
libSDL.so: /usr/lib/x86_64-linux-gnu/libSDL.so
```

I've verified that the following packages have been installed:
```
libsdl2-dev
libsdl1.2-dev
libsdl-ocaml
libsdl-ocaml-dev
```

I've also found the gitter discussion of June 14th this year and verified the additional dependencies mentioned there. If Ubuntu 22 is not supported yet, then this would be a good feature for the next release ;).

Cheers,
Tom

tomvand, 2022-09-02:
Ok, I managed to get it working. I've included the steps below. I don't know if every step was absolutely necessary, so consider these only for reference:

---

Remove apt's OCaml installation to prevent version conflicts with the manual install:
```
sudo apt remove *ocaml*
```

Install opam package manager for OCaml:
```
sudo apt install opam
opam init
eval $(opam env)
```

Switch to the latest OCaml version
```
opam switch create 4.14.0
eval $(opam env --switch=4.14.0)
```

Install dependencies except ivy
```
opam install ocamlbuild ocamlsdl2 xml-light ocamlnet conf-gnutls lablgtk camlp4 dune pcre ocamlsdl
eval $(opam env)
```

Install ivy from source

1. Download master here: https://gitpub.recherche.enac.fr/ivy-ocaml/snapshot/ivy-ocaml-9cb0e7768b7e0281a38686088cedeaee4d74790a.zip 
2. Extract, then run `make`, `make install`

---

Download all submodules (possibly only libopencm3 really necessary...)
```
git submodule update --init --recursive
```

Fix libopencm3 Makefile error. In file sw/ext/libopencm3/Makefile, replace line 28:
```
SRCLIBDIR:= $(subst $(space),\$(space),$(realpath lib))
```
with
```
SRCLIBDIR:= $(subst $(space),/$(space),$(realpath lib))
```
(note the front-slash `/` instead of back-slash `\`!) (Source: https://github.com/nanovna-v2/NanoVNA2-firmware/issues/73 )

Compile paparazzi, manually point ocaml to libSDL.so:
```
CAML_LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/:$CAML_LD_LIBRARY_PATH make
```

gautierhattenberger, 2022-09-02:
There are many issues with Ubuntu 22 that we have not solved yet, and SDL is one of them, so thank you for these tips, we will try to fix that (but after IMAV competition...)

tomvand, 2022-09-02:
Alright, hope it helps. I'll have a go at docker to see if that works, otherwise I'll reinstall an older Ubuntu.
Probably see you at IMAV ;)

Fabien-B, 2022-10-05:
Concerning the SDL issue, your last command is enough to make it work. No need to mess with opam or building ivy from sources.
An half-bad way to fix the issue is to add this line to the joystick Makefile:
```export CAML_LD_LIBRARY_PATH = /usr/lib/$(shell uname -m)-linux-gnu/```

It would be better is pkg-config could output this path, but I didn't managed to get it.

JustCurious8, 2022-11-08:
> Concerning the SDL issue, your last command is enough to make it work. No need to mess with opam or building ivy from sources. An half-bad way to fix the issue is to add this line to the joystick [...]

I managed to build the code using make for v6.1 version on ubuntu 22.04, but i am not able to build any nps and sim targets, compilation process for nps and sim targets are completing well with some const variable related warning during compilation but during the linking process of objects files, i am getting issues related to multiple definitions of variables and many first defined here .h files, which is causing errors and failing the build for any nps and sim targets. Is there a work around for this ? It's my first time building and getting into paparrazi UAV

Fabien-B, 2022-11-10:
I think it's due to the latest version of GCC which is not as permissive as the previous ones.
The code need to be fixed.
For a temporary fix, you can try removing some compilation flags to make GCC more permissive.

Fabien-B, 2022-11-16:
@JustCurious8 Most of the issues are now fixed in master. (#2950)
There is still the pending update of libopencm3, but you can fix the current version as stated above (replacing `\` by`/`):

Fix libopencm3 Makefile error. In file sw/ext/libopencm3/Makefile, replace line 28:
```
SRCLIBDIR:= $(subst $(space),\$(space),$(realpath lib))
```
with
```
SRCLIBDIR:= $(subst $(space),/$(space),$(realpath lib))
```

Fabien-B, 2023-01-24:
Closing this issue, see #2922 for the libopencm3 issue.

closed 2023-01-24 by Fabien-B