paparazzi/paparazzi #2911: Test build fails

fvantienen, 2022-10-03:
There is something wrong the rust key_generator which is unable to fetch packages.
```
Using "arch/stm32/px4fmu_2.4.ld" as ldscript for target "ap".
GENERATE KEYS
    Updating crates.io index
error: failed to get `rand` as a dependency of package `key_generator v0.1.0 (/home/pprz/paparazzi/sw/ext/key_generator)`

Caused by:
  failed to load source for dependency `rand`

Caused by:
  Unable to update registry `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to fetch `https://github.com/rust-lang/crates.io-index`

Caused by:
  error reading from the zlib stream; class=Zlib (5)
Makefile.ac:189: recipe for target 'generate_keys' failed
```

podhrmic, 2022-10-05:
Looks like this is a bug in an old cargo version. Updating cargo to 1.65 fixes it. If travis is pulling cargo from Ubuntu Bionic that is an ancient version. Maybe running `rustup update` would be enough? (but we would need to install [rustup](https://rustup.rs/) in each run).

fvantienen, 2022-10-07:
The change to go to ubuntu 20.04 didn't fix the problem

Fabien-B, 2022-10-07:
It was still a good thing to update to 20.04, we will do the rustup thing then...

fvantienen, 2022-10-20:
Can the rustup be added and tested? Or else for now it might be better to just remove the airframes from the CI build

Fabien-B, 2022-10-27:
Fixed

closed 2022-10-27 by Fabien-B