paparazzi/paparazzi #2932: Issues compiling NPS on Ubuntu 22.04

jesusBV20, 2022-10-19:
I'm testing on Ubuntu 22.04 the changes made by @Fabien-B in #2922 and I experienced a problem. Since I believe it is not directly related to the changes he has made in this PR, I decided to open this issue.

When I try to compile any airframe with NPS as the target, the compiler drops some multiple-definition errors like this:

/usr/bin/ld: 
/home/ubuntu-22/Workspace/paparazzi/var/aircrafts/MYAC/nps/nps/nps_ivy.o:
/home/ubuntu-22/Workspace/paparazzi/sw/simulator/nps/nps_main.h:28: 
multiple definition of `fdm_mutex'; 
/home/ubuntu-22/Workspace/paparazzi/var/aircrafts/MYAC/nps/nps/nps_flightgear.o:
/home/ubuntu-22/Workspace/paparazzi/sw/simulator/nps/nps_main.h:28: 
first defined here

Could It be related to a compiler flag?

gautierhattenberger, 2022-10-20:
I think it is a real bug, the fdm_mutex is not an extern variable. Not sure why it was compiling (or even working). Maybe the compiler of 22.04 is more strict than before.
I will fix that.

gautierhattenberger, 2022-10-20:
@jesusBV20 you can check with #2933

commit fafe256c (2022-10-20): [nps] variables in nps_main should be extern

fix #2932
PR #2933 (2022-10-21): [nps] variables in nps_main should be extern
fix #2932
commit 0b1ce467 (2022-10-21): [nps] variables in nps_main should be extern (#2933)

fix #2932
closed 2022-10-21 by fvantienen