paparazzi/paparazzi #2952: NavSegment routine bug

EwoudSmeur, 2022-11-14:
When using the path or go (with hmode route) instructions, I get a strange behavior when moving the starting waypoint while flying the segment. It seems to me that the line NavSegment routine is broken: if I change the starting waypoint while the drone flies the segment, instead of going along the new line it approaches a point next to the 'end' waypoint, after which it does fly to the end waypoint. As long as I don't move the waypoints during the flying of a segment it seems to work fine. Path or go with hmode route have the same behavior.

Tested only in NPS with a bebop simulation.

gautierhattenberger, 2022-11-17:
I have tested the segment navigation in simulation with the bebop2 example and the basic rotorcraft flight plan. I have not seen anything strange. If I move one of the WP (start or end), the A/C is trying to reach the segment before continuing in the right direction to the end of the segment. The trajectory is kind of "tangent" to the segment due to the position of the carrot moving along the line.
Do you have a schematic or a screenshot to explain the problem, because I'm not sure I getting it correctly.

EwoudSmeur, 2022-11-21:
It still occurs, so there is something wrong with that PR :(

closed 2022-11-21 by EwoudSmeur

EwoudSmeur, 2022-11-21:
I thought about it, and actually the behavior is exactly what I designed some years ago, it's just that I forgot about it.... To allow tracking of lines without a carrot the hybrid indi guidance has a function that basically generates a vector field that asymptotically guides the aircraft to a line. Therefore, if you move the beginning of the line, the drone will asymptotically approach the new line, but if it was on the line headed for the endpoint before, after moving the start point it will again asymptotically approach the line, which may mean it will not go through the end waypoint. Mystery solved 😅