paparazzi/paparazzi #2987: Release 6.2 does not work under Unbuntu 18.04 anymore

dewagter, 2023-02-11:
When installing paparazzi release 6.2 on a fresh ubuntu 18.04, it does not even start paparazzi center

 - missing lxml (sudo apt-get install python3-lxml)
 - missing numpy (sudo pip3 install numpy)
 - missing PyQt5 (sudo apt-get install python3-pyqt5)
 - missing dataclasses (sudo pip3 install dataclasses)
 - syntax error: supervision/python/paparazzicenter.py utils.py line 21: def remove_prefix(string: str, prefix: str, ,/) -> str
 does not understand this /) -> removed the ", /"
 - missing __future__ annotations: (tried: git clone -b legacy_py3.6 http://github.com/QUVA-Lab/e2cnn.git && cd e2cnn && sudo python3 setup.py install, but required python3.8)
 - unable to fix: recommend minimum python3.7
 - sudo add-apt-repository ppa:deadsnakes/ppa
 - sudo apt-get update
 - apt list | grep python3.8
 - sudo apt-get install python3.8
 - sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.6 1
 - sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.8 2
 - sudo update-alternatives --config python3 (and then type 2)
 - check python3 -V
 - sudo apt-get remove gpsd (gives errors with pkg)
 - missing PyQt5.Widget etc....

Tried removing old pythons
 - sudo apt purge python2.7-minimal
 - sudo apt purge python3.6
 - sudo ln -s /usr/bin/python2 /usr/bin/python
 - sudo apt install -y python3-pip
 - sudo ln -s /usr/bin/pip3 /usr/bin/pip
 - sudo apt autoremove
 - ls -al /usr/bin | grep python

Fabien-B, 2023-02-11:
Ubuntu 18.04 is being rolled out of support in 2 months.
I think it's not worth fixing, knowing that you can use the old paparazzi center.
Just launch it with `./paparazzi -legacy`

gautierhattenberger, 2023-02-11:
The `paparazzi-dev` package have not been updated in include the python libraries for the new pprz center. I also don't think it is worthy to change that. We might test for the ubuntu version to decide if the legacy version should be used by default or not.

dewagter, 2023-02-11:
Updated documentation so people know what to do and the start.py still works like before.

closed 2023-02-11 by dewagter

dewagter, 2023-02-12:
In summary: solving the python issues seems very challenging but as suggested by @gautierhattenberger the legacy mode still works great. This is now reflected in the documentation so this issue is resolved.