paparazzi/paparazzi #2988: Release 6.2 bebop-gazebo does not work with ubuntu22.04

dewagter, 2023-02-11:
- [x] compile ap https://github.com/paparazzi/paparazzi/pull/2986
 - [ ] nps_gazebo does not compile
 - [ ] gazebo9 -> gazebo https://github.com/tudelft/coursePaparazzi/issues/9
 - [ ] lot of Gdk-CRITICAL errors on make

dewagter, 2023-02-11:
Error:

CXX /home/mavlab/paparazzi/var/aircrafts/bebop_orange_avoid/nps/nps/nps_fdm_gazebo.o
In file included from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Spline.hh:22,
from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gazebo-11/gazebo/common/Animation.hh:22,
from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gazebo-11/gazebo/common/common.hh:5,
from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gazebo-11/gazebo/gazebo_core.hh:19,
from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gazebo-11/gazebo/gazebo.hh:20,
from /home/mavlab/paparazzi/sw/simulator/nps/nps_fdm_gazebo.cpp:35:
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Helpers.hh:40:13: warning: variable templates only available with ‘-std=c++14’ or ‘-std=gnu++14’
40 | constexpr T IGN_MASSMATRIX3_DEFAULT_TOLERANCE = T(10);
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Helpers.hh:990:28: error: ‘chrono_literals’ is not a namespace-name
990 | using namespace std::chrono_literals;
| ^~~~~~~~~~~~~~~
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Helpers.hh:1032:28: error: ‘chrono_literals’ is not a namespace-name
1032 | using namespace std::chrono_literals;
| ^~~~~~~~~~~~~~~
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/ignition/math6/ignition/math/Helpers.hh:1033:56: error: unable to find numeric literal operator ‘operator""s’

gautierhattenberger, 2023-02-11:
Gazebo is not part of the dependencies of `paparazzi-dev`, so it seems that it is only in the doc that we should update the installation instructions.

dewagter, 2023-02-11:
Correct. But nps_gazebo is part of paparazzi and does not compile anymore, that is what is **blocking**.
@OpenUAS ?

dewagter, 2023-02-12:
Gazebo11 requires C++17 or above.

dewagter, 2023-02-16:
https://github.com/paparazzi/paparazzi/pull/2992 solves this issue.

closed 2023-02-16 by dewagter