paparazzi/paparazzi #2989: Ubuntu 22.04 OCAML warnings on make

dewagter, 2023-02-11:
e.g.: 

File "server.ml", line 611, characters 10-26:
611 |   ignore (Glib.Timeout.add 1000 (fun _ -> update (); true));
                ^^^^^^^^^^^^^^^^
Warning 6 [labels-omitted]: labels ms, callback were omitted in the application of this function.
File "server.ml", line 639, characters 2-18:
639 |   Glib.Timeout.add period (fun () -> cb (); true)
        ^^^^^^^^^^^^^^^^
Warning 6 [labels-omitted]: labels ms, callback were omitted in the application of this function.
File "server.ml", line 976, characters 10-26:
976 |   ignore (Glib.Timeout.add 1000 (fun () -> periodic_handle_intruders (); true));
                ^^^^^^^^^^^^^^^^
Warning 6 [labels-omitted]: labels ms, callback were omitted in the application of this function.
OL server

Fabien-B, 2023-02-11:
This is currently being addressed in  #2856.
Should be merged next week.

Fabien-B, 2023-02-15:
Fixed in #2856

closed 2023-02-15 by Fabien-B