paparazzi/paparazzi #3022: DMAs not implemented on UART port

alessandro-mancinelli, 2023-04-08:
I noticed that the UART communication is not implemented using DMA (Holybro Pixhawk 4 with chibiOS).
This makes UART usage extremely processor intensive.
I'm currently running a setup that requires multiple high bitrate UART connections. This saturates the processor computational load to a point where just the UART threads are using 40% of my CPU power.
Does anyone have the knowledge of DMA UART implementation for this flight computer and OS?
Even just one or two DMA UART could make a huge improvement to my setup.

Thanks in advance to anyone that will be willing to help me (and all the future ppz users)! :)

gautierhattenberger, 2023-04-11:
ChibiOS has two drivers for UART communication, one with DMA and one without (called SD for SerialDevice if I'm not wrong). At the moment, the PPRZ uart driver is only implementing the second one. It is probably possible to do something to select the an other driver, or to eventually implement a 'uart_dma_arch' if configuration and usage is too different.
As a side note, using DMA is not always wanted or easy to implement due to the constraints on DMA channels with F4 and F7 boards. I think the DMA on H7 is making things much easier.

alessandro-mancinelli, 2023-04-12:
Thanks Gautier for the reply.
Perhaps I think it might be enough to create an if configuration statement to relieve the processor from one or two UART threads.
I know implementing DMA channels in F4 and F7 processors is a tedious task. I never had to solve the DMA channel puzzle with paparazzi.  Do you think it could be relatively easy to connect a DMA channel on Telem 1 to Telem 2 or the Pixhawk's UART port pin? Are there any DMA channels available right now?

gautierhattenberger, 2023-04-12:
I have never used a Pixhawk board and I have not made the configuration for this board, so I have no idea. If PX4 or Ardupilot are doing it, I guess it is possible, but for sure it is a pain to cross check all DMA stream and channels...
Again, checking the configuration of Ardupilot, might save you some time.

alessandro-mancinelli, 2023-06-19:
> ChibiOS has two drivers for UART communication, one with DMA and one without (called SD for SerialDevice if I'm not wrong). At the moment, the PPRZ uart driver is only implementing the second one. [...]

Hi Gautier! 
Following your suggestion I just moved to the cube orange (H7 processor). How complicated it is to move towards UART DMAs driver? Is it a human task or it requires deep knowledge of the system? Thanks a lot in advance!