paparazzi/paparazzi #3093: Rotwing in master

dewagter, 2023-09-21:
- [x] servos https://github.com/paparazzi/paparazzi/pull/3073
- [x] uavcan airspeed https://github.com/paparazzi/paparazzi/pull/3074
- [x] hanging rm3100 https://github.com/paparazzi/paparazzi/pull/3080
- [x] multi imu sending https://github.com/paparazzi/paparazzi/pull/3084
- [x] mavlink camera control https://github.com/paparazzi/paparazzi/pull/3085
- [x] moving target gains in airframe https://github.com/paparazzi/paparazzi/pull/3086
- [x] cube+ https://github.com/paparazzi/paparazzi/pull/3087
- [ ]  preflight checklist https://github.com/paparazzi/paparazzi/pull/2951
- [ ] sys_ident update https://github.com/paparazzi/paparazzi/pull/3075
- [ ] ground detect https://github.com/paparazzi/paparazzi/pull/3077
- [ ] wind speed ekf https://github.com/paparazzi/paparazzi/pull/3078
- [ ] airborne disarm https://github.com/paparazzi/paparazzi/pull/3081
- [ ] uavcan esc temp https://github.com/paparazzi/paparazzi/pull/3088
- [ ] actuator ID https://github.com/paparazzi/paparazzi/pull/3089
- [ ] multiple airspeed
- [ ] stabilization rotwing
- [ ] guidance rotwing
- [ ] transition rotwing

tmldeponti, 2023-11-14:
1. Add rotorcraft "type" instead of using guidance_indi_hybrid quad plane or tailsitter. Like that also other people that do not use guidance indi hybrid can use your code. 
2. Specify dependency of wing_rotation_controller on rotwing_state. 
3. In rotwing state remove dependecies on stabilizatuion_indi and guidance_indi_hybrid. This way other people that do not use these modules can still use the rotwing_state. For example the function "guidance_indi_hybrid_set_wls_settings" could be included in guidance_indi_hybrid.c

dewagter, 2023-12-17:
Completed

closed 2023-12-17 by dewagter