paparazzi/paparazzi #3095: covery warnings geofence

dewagter, 2023-09-21:
The following line causes a lot of warnings. Could it only print if the geofence was too low?

https://github.com/paparazzi/paparazzi/blob/ca1c1dbe8f97e7787b72327067665ec909fd7c49/sw/tools/generators/gen_flight_plan.ml#L1068

gautierhattenberger, 2023-09-21:
There is already a lot of error cases when geofence is too low. A "notice" would be better than a "warning" here, but as long as it is for `stderr` it will be reported in tests.

dewagter, 2023-09-22:
I am totally fine with printing it when the geofence was set too low. That is an actual warning. But on a good flightplan, this should be an info message, and is then not shown in ```make test```

Addressed in https://github.com/paparazzi/paparazzi/pull/3097

closed 2023-09-22 by dewagter