paparazzi/paparazzi #3237: Merge Oneloop with rotwing-state and scheduling

dewagter, 2024-02-12:
- [ ] #3210
 - [ ] #3214

tmldeponti, 2024-04-09:
- [x] Oneloop ANDI-INDI switch from actuator_pprz to commands
- [x] Jsbsim NPS adaptation to use commands instead of actuator_pprz
- [ ] RW3C airframe with commands instead of actuator pprz
- [ ] Rot wing state cleared from Guidance and stabilization INDI references
- [ ] Rot wing state, cleaner pitch scheduling definition. 
- [ ] Made clear what the actuator order is (see airframe)

tmldeponti, 2024-04-09:
To Do @dewagter @Dennis-Wijngaarden 

- [ ] Make sure to move all guidance code to guidance, all stabilization code to stabilization
- [ ] Switch from actuator_pprz to commands in your airframe and INDI
- [ ] Switch to generated autopilot
- [ ]

closed 2026-10-19 by tmldeponti