paparazzi/paparazzi #3244: ocamlc: unknown option `-safe-string'.

zzfzp, 2024-02-21:
During the make process, the (ocamlc: unknown option ` - safe string '.) that appears,Why is this?

1）
make -C sw/ext pprzlink.update
make[1]: Entering directory `/home/aston/paparazzi/sw/ext/pprzlink'
make[2]: Entering directory `/home/aston/paparazzi/sw/ext/pprzlink/lib/v2.0/ocaml'
Check for already installed pprzlink ocaml lib
Build bytecode lib
+ ocamlfind ocamlc -safe-string -c common/convert.c
ocamlc: unknown option `-safe-string'.

2）aston@ubuntu:~/paparazzi$ ./paparazzi
  File "/home/aston/paparazzi/sw/supervision/python/paparazzicenter.py", line 36
    self.conf: Conf = None
             ^
SyntaxError: invalid syntax

Fabien-B, 2024-02-21:
The two errors seems unrelated to each other.
1) on what OS (and version) are you ? What is your version of ocamlc ? (`ocamlc --version`)

2) That's curious as well, what is your version of python ? (`python3 --version`)

OpenUAS, 2024-05-21:
@zzfzp for now the issue closed since you gave no response to the request for more information to help you to fix your issue. No hard feelings, feel free to re-open this issue if you still have not found a solution, maybe someone can then be of more help.