paparazzi/paparazzi #3252: Crazyflie ChibiOS firmware crashing

Pelochus, 2024-03-19:
## The problem
Crazyflie 2.1 does not work. All LEDs except one are lit without heartbeat (there are 2 red LEDs, 2 green LEDs and 2 blue LEDs. If I'm not mistaken, the blue LED corresponding to the system timing is the one turned off)

My supervisor @noether suggested me to open an issue here mentioning @Fabien-B. I would also thank anyone trying to flash the firmware to a Crazyflie they have lying around to see if the problem is on my end and not in Paparazzi or ChibiOS.

## What I tried
Flashing the firmware on one Crazyflie with Debian 12 and Ubuntu 22, two different laptops. In both laptops I tried ap, test_gpio, test_sys_time and test_led. I also tried the profile from ```conf/airframes/test_boards``` and ```conf/airframes/OPENUAS```.  

## What I found out
I recently debugged with a STLink V2 and OpenOCD the Crazyflie with ```test_led``` and ```ap``` firmwares flashed.

- In the case of ```test_led``` the Crazyflie ends up stuck in an infinite loop in the function ```idle_thread``` [(link)](https://chibiforge.org/doc/21.11/full_rm/group__instances.html#ga0ea7aefc26b9e8ff69c343baa2acb0ef) from ChibiOS (```chinstances.c``` file) 
- For ```ap```, it ends up in a ```unhandled_exception``` function

In both cases, OpenOCD ```reset halt``` command does not work, so I can only either reset the board or halt the board manually after a reset, so I can't trace how both firmwares end up in those functions.

## Will try soon
- Try to fix OpenOCD/GDB for actual, proper debugging
- Debug test_sys_time
- Debug test_gpio
- Try to compile the firmware with Paparazzi around this commit (seems to be the most recent Crazyflie commit that seems to still work): https://github.com/paparazzi/paparazzi/commit/45b020d678b918a74c00d407100faa83930da40d 

### Side note 1
I've managed to install Paparazzi on Debian 12 with some minor adaptations. Should I open a new issue or PR in order to help porting Paparazzi to Debian 12 (could also work for Debian 11)?

### Side note 2
The Python script for the Crazyradio needs ```cflib``` . There is little documentation about this, I believe there should be a little more information elsewhere about how to use the Crazyflie. Just a minor suggestion I think.

Pelochus, 2024-03-23:
Good news, yesterday I found out why ```test_led``` didn't work. At least now I have heartbeat on the Crazyflie.
In this file [```chibios_test_led.c```](https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/test/chibios_test_led.c) I changed this:

```c
static THD_WORKING_AREA(waThread1, 256);
static THD_FUNCTION(Thread1, arg) {

  (void)arg;
  chRegSetThreadName("blinker");
  while (true) {

    //palSetPad(GPIOD, 13);       /* Orange.  */
    palSetPad(GPIOB, 12);       /* Orange.  */
    LED_ON(1);
    chThdSleepMilliseconds(500);
    //palClearPad(GPIOD, 13);     /* Orange.  */
    palClearPad(GPIOB, 12);     /* Orange.  */
    LED_OFF(1);
    chThdSleepMilliseconds(500);
  }
}
```

To this (just changed from GPIOB to C and from 12 to LED_RED_R):

```c
static THD_WORKING_AREA(waThread1, 256);
static THD_FUNCTION(Thread1, arg) {

  (void)arg;
  chRegSetThreadName("blinker");
  while (true) {

    //palSetPad(GPIOD, 13);       /* Orange.  */
    palSetPad(GPIOC, LED_RED_R);       /* Orange.  */
    LED_ON(1);
    chThdSleepMilliseconds(500);
    //palClearPad(GPIOD, 13);     /* Orange.  */
    palSetPad(GPIOC, LED_RED_R);     /* Orange.  */
    LED_OFF(1);
    chThdSleepMilliseconds(500);
  }
}
```

Which made me think if ```test_led``` works on every device, since it does not seem as a generic file for every device.

gautierhattenberger, 2024-03-23:
The `RTOS` missing is clearly  a mistake. The funny thing is that it actually works for me. The autopilot is running and I have datalink with the ground station. So it means that the AP can run in the idle task, which is obviously not what we want. I will retry with the fix to use the correct main file.
Regarding the test led program, since it is used by all boards, it is hard to make it generic unless the led names are standardised. But at least it is good news that it works.

Pelochus, 2024-03-23:
Hi @gautierhattenberger thanks for your response. It is interesting that it actually works without the ```RTOS``` var in the makefile. From what I've debugged until now, it seems that without it there is at least more functionality while debugging without the ```RTOS``` set in the makefile. Let me know if you discovered something on your end, perhaps there are some abandoned commits (related to the makefile and firmware) that were never pushed from local?

I will soon try to debug why there is no heartbeat in the ```ap``` firmware, the only thing I discovered until now is what I've mentioned, ```ap``` firmware has 6 threads which 2 are dedicated to the UART6/nRF which only execute once. The rest of the threads seem to do something (or try to), but none of them toggle the LED (or so it seems).

Regarding the ```test_led```, perhaps we could implement a brute-force way like this?

```c
// Let's suppose GPIO banks from A to D are represented with ints from 0 to 3
for (int gpio = 0; gpio < 4; gpio++ {
    for (int pin = 0; pin < 32; pin++ {
        palSetPad(gpio, i);
        LED_ON(1);    
    }
    
    chThdSleepMilliseconds(500);

    for (int pin = 0; pin < 32; pin++ {
        palSetPad(gpio, i);
        LED_OFF(1);    
    }
    
    chThdSleepMilliseconds(500);
}
```

This could work on many boards, I believe. Of course there are better ways but most of them I can think of are harder.

gautierhattenberger, 2024-03-25:
@Pelochus see #3255 for a fix. At least for me the firmware is running and the leds are flashing. Attitude is okay, but not tested in flight.

Pelochus, 2024-03-25:
Just tried the fix. LEDs are working now (finally!) but still no radio communication. The radio seems connected, but every packet sent is 0 bytes in size. I will investigate it a bit more and report back soon.

Pelochus, 2024-03-25:
So it seems that ChibiOS does exactly one context switch to the threads controlling the radio, so no information is sent. The radio DOES receive packets, however, they are totally empty since the STM32 does not communicate with the radio chip and seems that the radio wants to send empty packets if it doesn't receive data from the main processor. 

@gautierhattenberger my workflow right now is the following:

- Compile and flash the firmware with your new changes, based on the default configuration from ```conf/test_boards.xml```
- Commented those blocking modules in the XML since I don't use them
- Connect to the Crazyflie with the Crazyradio2 python script
- Check with the Paparazzi messages tool that there are no packages received 

Are you including any extra steps apart from those? Perhaps I need to flash some new firmware (or an older one) to the radio?

gautierhattenberger, 2024-03-25:
I have tested again with the following setup:
- Ubuntu 22.04
- Paparazzi with the branch in  #3255 (and the 4 unwanted modules commented from demo file)
- Crazyradio hardware Rev.C, `cflib` installed from pip a few days ago
- CrazyFlie v2.1, flashed with DFU mode (using `dfutil`)
- `crazyradio2ivy.py` without any arguments + usual tools (server, gcs, messages)

Everything is working for me, tested on 2 different crazyflie. Only thing I have not tried is to start the motors as I don't have a RC at the moment. So it is hard to tell what is your problem as I can't reproduce it. Your 1st screenshot just tells that you don't have RC input (I don't know if you are sending RC commands other telemetry or not). The second is not clear. What is happening ? The function is not called or is called but does nothing ?

Pelochus, 2024-03-25:
Thanks for sending your setup. I'm in fact not using any RC controller right now, I assumed that the Crazyflie should report back with telemetry (altitude, angles, speed etc...), isn't that right? 

What I tried to mean with the 2nd screenshot is that the behaviour of the functions ```syslink_dl_event()``` and ```pprz_dl_event()``` do not return any information back (basically, they skip their instructions because the first if statement in them is false, 0 or whatever), similarl to the 1st screenshot.

My current setup is nearly identical to yours, just changes the Crazyradio that is hardware rev F.

If you don't mind, could you send me a .zip file with the contents of ```var/aircrafts/CRAZYFLIE```? I could try flashing that firmware just to discard that the problem is either in Paparazzi Center/GCS, the ```crazyradio2ivy.py``` script or the Crazyradio/Crazyflie hardware (we also have 2 Crazyflies, so it shouldn't be that, but just to be safe).

gautierhattenberger, 2024-03-26:
#3255 only solves a part of the problem, so this issue is still opened.

Pelochus, 2024-03-27:
Just tried the firmware provided by @gautierhattenberger. Still no messages sent from the Crazyflie to the Crazyradio. Tried to debug the python script just in case it was Paparazzi's fault and definitely it is not, the radio in fact receives empty packets (or so it says the python script).

Just to make sure, the default behaviour of the Crazyflie sends some telemetry back periodically right (speeds, accelerations etc...)?

I'm going to try now to revert back to an older official version of the firmware by Bitcraze to see if the nRF chip firmware has changed in recent updates so that it is not compatible with the current implementation in Paparazzi.

Pelochus, 2024-03-27:
Tried older official firmwares from 2019.02 to 2020.04 and then flashed over the Paparazzi firmware. I used these because they are dated from around when the first Paparazzi implementation of Crazyflie was merged. 
- The LED corresponding to the battery now works and flashes when the battery is connected, so that seems good news
- There is also context switches to the UART6 threads in ChibiOS. Reminder that these threads correspond to the datalink with the radio. **Note: I find interesting that the RX thread is executed _way_ more times than the TX one, perhaps the firmware is waiting for an initial message or something?**
- Still same behaviour with the radio, receives empty packets
-  Tried (just in case) if the radio works with the older official firmware, since the radio I am using (Crazyradio 2.0) was released around 2023, way after the Crazyflie 2.1 Paparazzi firmware was developed. I suspected that the problem could be here (Paparazzi implementation is adapted to older nRF firmware and/or radio HW/firmware) but doesn't seems to, since the official older firmware (2019.09) works with the newest radio firmware (around 2023)

Right now, I have discarded (or almost fully discarded) the following options/scenarios:
- ChibiOS now seems to be fixed
- Paparazzi Center and related tools are not the problem, since we have tested on both Debian 12 and Ubuntu 22.04, different releases, on Ubuntu 20 with even older commits etc...
-  We have 2 Crazyflies, so not a hardware problem (we still need to try recent firmwares on the other Crazyflie though, just in case, but until now the behaviour for same scenarios was the same)
- Probably not the firmware for the STM32 chip from Paparazzi, since it works for @gautierhattenberger and using his compiled firmware for our Crazyflies results in same behaviour compared to our compiled firmware
- Similarly, the nRF chip firmware does not seem to be the problem. After returning to older versions of it, the modern radio firmware stills connect to it and the behaviour after flashing Paparazzi's autopilot is the same. As far as I'm concerned, Paparazzi does not change this firmware right? 
- The Python script for the radio should also not be a problem, works for @gautierhattenberger and we have tested (again) under both Debian and Ubuntu. It receives packets after all, just that they are empty.

All that said, I would appreciate if anyone has any idea where I could investigate more or if I am missing something, like (to put an example) some configuration in Paparazzi that makes the Crazyflie send telemetry back to the ground station. Right now, like I said in previous comment, I am starting the Crazyflie, connecting the radio and opening a messages window in Paparazzi. It always receives empty packets, that can only be seen using ```print()``` in the Python script for the Crazyradio. It never reaches the ivybus.

gautierhattenberger, 2024-03-27:
You said that you only run the `crazyradio` script and the `messages` tool on ground. But have you tried to run the full session with `server` and `GCS` as well ?

Pelochus, 2024-03-27:
Yes I tried, but as you said, it shouldn't matter. This also confirms that I don't need to do some extra config in Paparazzi, I'm running out of ideas...

Pelochus, 2024-03-28:
I just found out a VERY interesting discovery. The python script for the Crazyradio doesn't receive data from the official firmware, but it still connects to it. The official Crazyflie Client does actually receive information (battery, IMU etc...) from the Crazyflie. 

Since the nRF and Crazyradio firmware are exactly the same and the python script uses the official ```cflib``` from Bitcraze, I thought the script could receive data from the official firmware. To my surprise, it has the same behaviour as usual: empty packets. This behaviour may be due to driver/hardware/firmware changes in the Crazyradio 2.0 compared to the Crazyradio PA, so the current script probably needs some adaptations.

I'll take a look into the documentation for ```cflib``` soon, but before that @gautierhattenberger I'd like you to confirm that you were using the ```crazyradio2ivy.py```script as in current ```master``` and that your cflib version is ```0.1.25.1``` (I want to make sure before diving into Python, I am not very used to that language).

I will also ask to the Bitcraze team, they might give me a hand developing the new script.

gautierhattenberger, 2024-03-28:
Yes I confirm this is my `cflib` version with the `crazyradio2ivy.py` on master. And I'm using python 3.10.12 from ubuntu.
It is indeed possible that your radio dongle is the source of the problem since this is the only hardware part that differs from mine.

Pelochus, 2024-04-02:
Fixed finally! Surprisingly, even though ```print(pk.port)``` said otherwise, I shouldn't have changed ```CRTP_PORT_PPRZLINK = 9``` to 15. I changed that back to 9 and it works! Now, I'm going to test newer official firmwares to see if that has to do something with the problem, since, if I'm not mistaken, I had tried that solution various times before... Weird.

Btw @gautierhattenberger do you remember what's the purpose of that variable?

gautierhattenberger, 2024-04-02:
Hum, clearly you shouldn't change the port number... Have you seen somewhere that this should be changed ? Port 9 is one of the port available for custom application, so basically cflib will just forward the data without looking at it (I guess). Ports defined on airborne side and ground side should be the same. Port 15 is already used by bitcraze.

The airframe file with the modules is also used for compilation tests, so maybe add a new airframe example (or make an i2c blocking function with timeout...)

Pelochus, 2024-04-04:
I just debugged what the value for pk.port was, and since it always printed 15, I thought it varied from Crazyflie to Crazyflie. Seems it was a problem before fixing ChibiOS and flashing the older firmware. After all, the Crazyflie seems to be working, have not tested yet the motors. In short the solution was this:

1. Fixing the pin definitions (done by you guys)
2. Flash an older firmware from Bitcraze (2019.09 works) before flashing Paparazzi AP
3. Comment or remove blocking modules if using stock Crazyflie 2.1 hardware configuration

closed 2024-04-04 by Pelochus