paparazzi/paparazzi #3264: Crazyflie goes into AP_KILL mode after Start Engine from GCS

Pelochus, 2024-04-11:
## Issue
This is kind of the continuation of this issue #3252.

Crazyflie 2.1 doesn't completely start the engine after selecting the "Start Engine" option from the GCS. I'm not using a radio. In this screenshot you can see my configuration for compiling the autopilot. 

[image]

Which is, essentially, the default configuration but with a modified airframe file which removes external modules for the Crazyflie.

 ## Interesting findings
The motors do in fact start spinning, however, this can only be checked while debugging. We can discard a problem related to the GPIOs like in issue #3252. What I debugged goes like this:

- `modules_actuators_periodic_task()` starts (first time after hitting **Start Engine** on GCS)
- Motors start spinning slowly
- `modules_actuators_periodic_task()` comes again (second time)
- Motors stop
- Crazyflie goes into AP_KILL

Obviously, while trying this out while not debugging, the PWM signal stops so soon that the motors can't start spinning.

## Side note
Paparazzi NPS does not work correctly with Crazyflie (works but Z goes to infinity, making the Crazyflie go crazy (hehe)), perhaps this has to do something with the programming of the motors or something?

gautierhattenberger, 2024-04-12:
The KILL mode depends on battery level, gps status, RC status, configuration flags etc. To understand why it goes to kill mode you have to identify the conditions that triggers it (or give more info on the points I mentioned).
Hint for the debugger: make a breakpoint on the `autopilot_static_set_mode` function; when called with `AP_MODE_KILL`, analyse the call stack to see the triggering condition.

Pelochus, 2024-04-12:
Thanks Gautier. I'll report back with extra info soon (I don't have the Crazyflie right now)

Pelochus, 2024-04-15:
For now I discovered that that the autopilot is set to FAILSAFE_MODE and then this happens:

[image]

I'm going to check now why the hell does it go to FAILSAFE...

Pelochus, 2024-04-15:
So this calls it: 

[image]

`USE_GPS` is set to true, which I believe it shouldn't due to in `conf/airframes/test_boards/crazyflie_2,1.xml`

[image]

Pelochus, 2024-04-15:
Got a temporal solution:

In first photo from previous comment, hardcode GpsIsLost() with 0. It doesn't work if changing "FALSE" to 0 in 3rd photo.

Any suggestion why this might happen?

Pelochus, 2024-04-15:
Doesn't solve it directly, I'm trying out new ideas for solving it without hardcoding, tomorrow I'll report back

gautierhattenberger, 2024-04-15:
This is a kind of side effect of recent changes on modules dependencies.
You are entering failsafe because you try to fly in nav without gps.
If you were using a gps or positioning system like optitrack, it would be just fine. If you want to fly without a module providing the "gps" functionality (and with the flag `NO_GPS_NEEDED_FOR_NAV` to disable checks), compilation will fail because is is required by the ins module. This was not the case when I made the support for crazyflie.
So the solution mostly depends on your setup and the type of flights you want to make:
- in direct attitude control with a joystick, it should work
- in nav mode, if you have a position from optitrack, gps or equivalent, use that
- in nav mode without position: be more specific about the type of flights and sensors you plan to use. The selection of modules will depend on that.

Pelochus, 2024-04-15:
Thanks Gautier. I'm going to use the Crazyflie with the official opticflow module from Bitcraze.
https://www.bitcraze.io/products/flow-deck-v2/

If I'm not mistaken Paparazzi has support for this right? What would you use in this scenario?

noether, 2024-04-16:
> Thanks Gautier. I'm going to use the Crazyflie with the official opticflow module from Bitcraze.
> https://www.bitcraze.io/products/flow-deck-v2/
> 
> If I'm not mistaken Paparazzi has support for this right? What would you use in this scenario?

And the altimeter.

Although we will add external localization in the near future, we would like to fly them with only onboard localization (not ready now either). 

For now, just optical flow and altimeter from the official deck. So we can move around in autonomous mode keeping the vertical and moving at least back and forth.

Thanks a los Gautier and guys for the support!

Pelochus, 2024-04-16:
Fixed!

In `airframe/test_boards/crazyflie_2.1.xml` I moved:

`<define name="NO_GPS_NEEDED_FOR_NAV" value="TRUE"/>`

Under AUTOPILOT section:

```xml
<section name="AUTOPILOT">
    <define name="NO_GPS_NEEDED_FOR_NAV" value="TRUE"/>
    <define name="MODE_STARTUP" value="AP_MODE_NAV"/>
    <define name="MODE_MANUAL" value="AP_MODE_ATTITUDE_DIRECT"/>
    <define name="MODE_AUTO1"  value="AP_MODE_ATTITUDE_Z_HOLD"/>
    <define name="MODE_AUTO2"  value="AP_MODE_NAV"/>
  </section>
```
Then on `autopilot_static.c`, from this:

```c
#if USE_GPS
#include "modules/gps/gps.h"
#else
#if NO_GPS_NEEDED_FOR_NAV
#define GpsIsLost() FALSE
#else
#define GpsIsLost() TRUE
#endif
#endif
```

To this:

```c
#include "generated/airframe.h"

#if USE_GPS
#include "modules/gps/gps.h"
#endif

#if NO_GPS_NEEDED_FOR_NAV
#define GpsIsLost() FALSE
#else
#define GpsIsLost() TRUE
#endif
```

The only problem here is that it would substitute this in `gps.h`:

```c
#define GpsFixValid() gps_fix_valid()
#if USE_GPS
#define GpsIsLost() !GpsFixValid()
#endif
```

However that can be solved either adding a new define in the Crazyflie airframe file or a more generic solution would be:

```c
#include "generated/airframe.h"

#if NO_GPS_NEEDED_FOR_NAV
#define GpsIsLost() FALSE
#else
#define GpsIsLost() TRUE
#endif

#if USE_GPS
#include "modules/gps/gps.h"
#endif
```

Basically, inverting the #if orders.

gautierhattenberger, 2024-04-16:
The main issue is that navigation and "autopilot" are looking for GPS status, rather than a state status. It is the state interface that should be improved in that regard.
Even without that, things should work without GPS (so when USE_GPS is not defined). Not sure sure it is the case at the moment because of the requirements from ins module.
The only INS that can deal with velocity without position is the EKF2 I think. My advice would be to select the correct parameters to use the opticflow deck, and update the functionalities required/provided by the modules to make the dependency checking happy.

Pelochus, 2024-04-18:
Okay, thanks Gautier. For now I'm going to try whether my fix works for what I want to do or not and then I'll try your suggestions if it doesn't. 

I'm closing the issue for now, I might reopen it in the future, but let's hope not

closed 2024-04-18 by Pelochus