paparazzi/paparazzi #3307: [settings] Add getters

Fabien-B, 2024-06-14:
In the settings, we have the `handler` that allows to call a function to set the value.
We need the same mecanism to get the value: a getter.

This getter should return a `float`, to be consistent with the current system.

## Current situation

Let's take an example with this setting:
```XML
<dl_setting var="autopilot.kill_throttle" min="0" step="1" max="1" module="autopilot" values="Resurrect|Kill" handler="KillThrottle"/>
```

```C
#define DlSetting(_idx, _value) { \
  switch (_idx) { \
     ...
    case 13: autopilot_KillThrottle( _value ); _value = autopilot.kill_throttle; break;\
    }\
}

#define PeriodicSendDlValue(_trans, _dev) { \
     ...
    case 13: var = autopilot.kill_throttle; break;\
    ...
}

static inline float settings_get_value(uint8_t i) {
  ...
  case 13: return autopilot.kill_throttle;
  ...
}
```

:warning: First of all even in the current system, I think `_value = autopilot.kill_throttle;` (line 4) should be removed: If there is a handler, the value should not be forcefully changed after calling the handler.

## Proposed new behavior

The setting could be changed like this:

```XML
<dl_setting name="KillThrottle" getter="getKillThrottle()" min="0" step="1" max="1" values="Resurrect|Kill" handler="KillThrottle"/>
```

It would result in this generated code:

```C
#define DlSetting(_idx, _value) { \
  switch (_idx) { \
     ...
    case 13: KillThrottle( _value ); break;\
    }\
}

#define PeriodicSendDlValue(_trans, _dev) { \
     ...
    case 13: var = getKillThrottle(); break;\
    ...
}

static inline float settings_get_value(uint8_t i) {
  ...
  case 13: return getKillThrottle();
  ...
}
```

Defining the getter like this `getter="getKillThrottle()"` allows to specify either a function call, or a global variable like in the current situation.

The `name` attribute would be added to define the name of the setting. `shortname` would be deprecated in favor of `name`.
The name resolution order would be: `name`, `shortname` (with deprecated warning), `var`.

## XML valid cases

```XML
<dl_setting var="autopilot.kill_throttle" min="0" step="1" max="1" values="Resurrect|Kill"/>
```

```XML
<dl_setting name="KillThrottle" var="autopilot.kill_throttle" min="0" step="1" max="1" values="Resurrect|Kill" handler="KillThrottle"/>
```

```XML
<dl_setting name="KillThrottle" getter="getKillThrottle()" min="0" step="1" max="1" values="Resurrect|Kill" handler="KillThrottle"/>
```

```XML
<dl_setting name="KillThrottle" var="autopilot.kill_throttle" getter="getKillThrottle()" min="0" step="1" max="1" values="Resurrect|Kill"/>
```

The following should Fail at compilation, because `var` would be unused:
```XML
<dl_setting var="autopilot.kill_throttle" getter="getKillThrottle()" min="0" step="1" max="1" values="Resurrect|Kill" handler="KillThrottle"/>
```

Fabien-B, 2024-06-18:
Hi, there are multiple times where a variable is not needed, but you need to define it anyway because of how settings works currently.
Here is some awful code I wrote recently: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/computer_vision/tag_tracking.h#L66-L76
Here the variable is useless: https://github.com/paparazzi/paparazzi/blob/36828c0560e9b57607b043dad028e39411775b30/sw/airborne/modules/loggers/sdlog_chibios/usbStorage.c#L171-L177

And in a more general way, the current system forces us to have global variables everywhere, which is a bad design choice.
Sometimes we access directly these variables instead of using the dedicated functions (or macro) that may do essential checks.
Also, it's very difficult to understand how paparazzi works because of these global variables.

We can't change everything in paparazzi at once, but a change like this can allow us to slowly shift towards cleaner code.

dewagter, 2024-06-19:
Thanks a lot. Great suggestion. I'll investigate which unnecessary variables we introduced once this is merged.