paparazzi/paparazzi #3378: NatNet not working anymore

dewagter, 2024-10-01:
[image]
That's the output of the natnet2ivy.py script with the -v flag up. It is not able to properly identify the stream coming from motive and shuts down the interface. Perhaps the pose packets are different in the old SDK. Could we somehow make a flag (like the current "-o" flag) to allow backward compatibility? 
Probably we could also solve by purchasing a motive 3.x license but it's nice to have the capability of rolling back and use motive 2.x for any reason.

_Originally posted by @alessandro-mancinelli in https://github.com/paparazzi/paparazzi/issues/3249#issuecomment-2379062679_

gautierhattenberger, 2024-10-02:
I agree that this is a problem. The natnet lib, especially in python, gives very little information about the possible issues and the differences between APIs. I don't think it even says that it should not be compatible.
One solution would be to analyze the errors in details and fix their code, if possible. Another option is just to have the two code side by side and select at startup the version you want. Since the callbacks prototypes and objects have changed, it still requires a little bit of work to make this not too ugly.

fvantienen, 2024-10-23:
https://github.com/tudelft/paparazzi/commit/5b6730dece7755567a2ec8aeb1b5910eb065ea11#diff-dd2eab9474ada20e59ef6082f6fdf7449109f2e934baf22624910e75299eede7R93
This is not the proper fix, but makes it work with the old natnet. If we add something like this properly would be nice

gautierhattenberger, 2024-10-24:
Here are the results:
- with `__nat_net_requested_version = [3,0,0,0]`, it is never working, it really needs to have `[0,0,0,0,]`. Even with that it is not starting every time, but this is a reported bug with this version, hopefully corrected with the "final" release of natnet coming soon.
- the `__server_version` works the same with `[3,0,0,0]` and `[0,0,0,0,]`
- we never have the `not natnet.connected()` case

closed 2026-03-06 by OpenUAS