paparazzi/paparazzi #3417: Does PaparazziUAV support Raspberry Pi?

Lqs66, 2024-12-20:
I would like to ask if PaparazziUAV's nps support runs on Raspberry Pi?

gautierhattenberger, 2024-12-20:
NPS is a mix C and C++, compiled with GCC, so it should run on a raspberry py as well. It requires the JSBSim library, with is available as a ubuntu package, including arm, or it can be compiled from source.

OpenUAS, 2025-06-11:
I remember we got in long past running the GCS and NPS on a Gumstix board see https://wiki.paparazziuav.org/wiki/GettingTheGCSRunningonAGumstixBoard 
Maybe it still helps.

Lqs66, 2025-06-16:
> I remember we got in long past running the GCS and NPS on a Gumstix board see https://wiki.paparazziuav.org/wiki/GettingTheGCSRunningonAGumstixBoard Maybe it still helps.

Thanks for the reply, I'll try it.

closed 2025-08-14 by OpenUAS