paparazzi/paparazzi #3454: IMU no longer working on Pixhawk4

OpenUAS, 2025-04-17:
Having a **Pixhawk4** and upload e.g. Splash4 airframe of branch v6.4 all OK, checkout master as of 20250417-2000CET (don't have git object sha at hand) no more attitude, IMU_LOST message. Tested with Exact same hardware on desk.

All other sensors AFAIK where still working, baro, airspeed, external mag, telemetry, rc, all fine 'n dandy.

gautierhattenberger, 2025-04-17:
I couldn't find changes that would explain this. The drivers for the pixhawk4 and the imu mpu6000 have not really changed since v6.4. I don't know if I can find such old sensor (and I don't have this autopilot). Can you test the hardware lines with a logical analyzer to see if there is something passing or nothing at all ?

OpenUAS, 2025-05-20:
Thx, indeed just tried
git checkout -b investigate_PixhawkIMU_nodata 5e9e61e499aaed0971933e7045e3e10244b418b0

No longer works
git checkout -b investigate_PixhawkIMU_nodata2 2a9887a74755e8372923a97f43333a43c242b02c

BTW if flashing via PX4 bootloader USB also change in  conf/boards/px4fmu_5.0_chibios.makefile

PX4_BL_PORT ?= "/dev/serial/by-id/usb-3D_Robotics_PX4_BL_FMU_v5.x_0-if00"  to PX4_BL_PORT ?= "/dev/serial/by-id/*BL*"

fvantienen, 2025-05-21:
See https://forum.chibios.org/viewtopic.php?f=35&t=6566 for the fix

PR #3457 (2025-05-21): [chibios] Fix SPI driver for STM32F7xx
This should fix #3454
closed 2025-05-21 by fvantienen