paparazzi/paparazzi #3493: Lisa/M 2.0 luftboot error

lucafedechen, 2025-07-02:
hi to all.using latest master i can't flash the lisa/M trought usb luftboot...try with old stm32_mem.py same error

run --verbose --dry-run get this

USB Device Firmware Upgrade - Host Utility -- version 1.3
Copyright (C) 2011  Black Sphere Technologies
Copyright (C) 2012  Transition Robotics Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

Found 2 DFU devices.
Found DFU device : ID 1d50:600f b'Transition Robotics Inc.' - b'Lisa/M (Upgrade) V1.1' - b'LM2-373 34FFD4053452323340281643\x12'
Whoops... could not get device description.
Could not find STM32 (autopilot) device.

[image]

[image]

[image]

OpenUAS, 2025-07-07:
OK, It's bee a while, can take a look this week, have somewhere still one Lisa M 2.0 for testing, the rest all upgraded to MX with F4.

REQ: A bit more info would be helpful, could you cut/paste this in terminal and pres enter then copy (CTRL+C) the output to this issue?
`uname -a && python --version`

lucafedechen, 2025-07-07:
hi @OpenUAS thanks for help me

result:

Linux lucaPC 6.8.0-60-generic #63~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 22 19:00:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
Python 3.10.12

also i try replace dfu.py and stm32_mem.py files from version 5.8 and 5.0 but issue not solve
thanks again

closed 2025-08-14 by OpenUAS