paparazzi/paparazzi #3504: Old UAVCAN manual decoding remaining

Fabien-B, 2025-07-11:
Hello,
I noticed in [can_fuelcell.c](https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/energy/can_fuelcell.c) that some old manual decoding (`canardDecodeScalar...`) is still there.
If this module is still used, could it be refactored by using DSDL definitions ?
If needed, a custom data type can be created : https://dronecan.github.io/Implementations/Libuavcan/Tutorials/8._Custom_data_types/

fvantienen, 2025-07-14:
This is still being used with this DSDL description and repository: https://github.com/tudelft/DSDL/blob/326c4e12b0fa92cec1070639ff14863bb32b0664/uavcan/equipment/fuelcell/1141.Status.uavcan

Fabien-B, 2025-07-15:
Should now be working with #3505.
Be careful that I change the ID to comply with the convention, so the fuelcell itself should be updated.
If needed we can use the old ID, but this would be contrary to the ID distribution convention.

closed 2025-07-15 by Fabien-B