paparazzi/paparazzi #3543: QMC5883L driver freezes after some time, requires reset to recover

alrochas1, 2025-09-29:
Hello,

We have observed that the QMC5883L driver in Paparazzi sometimes **freezes after a random period of time** (a few minutes to several hours).  When this happens, the magnetometer stops providing new data and remains stuck until the autopilot is rebooted.

During code analysis, I identified what appears to be a bug in the `qmc5883l.c` module:
```
mag->i2c_trans.buf[0] = QMC5883L_REG_RESET_PERIOD;
mag->i2c_trans.buf[0] = 0x01;   // overwrites buf[0] again
```

However, this issue seems unrelated to the freezing behavior. I suspect the root cause lies in the I2C driver's inability to handle temporary communication losses. Once the state machine enters a stuck state, it fails to recover autonomously.

### Temporal Fix

I added a simple watchdog that resets the driver state if it stays too long in `STATUS_MEAS` without data:
```
#define QMC5883L_WATCHDOG_LIMIT 100
uint8_t  q5883l_errors_counter = 0;
static uint16_t qmc5883l_watchdog_counter = 0;

void qmc5883l_watchdog(void)
{
  if (!mag_qmc5883l.initialized) {
    qmc5883l_watchdog_counter = 0;
    return;
  }

  if (mag_qmc5883l.data_available) {
    qmc5883l_watchdog_counter = 0;
    return;
  }

  if (mag_qmc5883l.status == QMC5883L_STATUS_MEAS) {
    qmc5883l_watchdog_counter++;
    if (qmc5883l_watchdog_counter > QMC5883L_WATCHDOG_LIMIT) {
      mag_qmc5883l.i2c_trans.status = I2CTransDone;
      mag_qmc5883l.status = QMC5883L_CONF_UNINIT;
      mag_qmc5883l.initialized = false;
      qmc5883l_watchdog_counter = 0;
      q5883l_errors_counter++;
    }
  }
}
```

gautierhattenberger, 2025-09-30:
If I got it correctly, the issue is in the mag driver (and its internal logic) rather than the I2C driver itself ?
Looking at this driver, I see that if the sensor is in the `STATUS_MEAS` state, it will only go back to idle if the transaction is a success. The `I2CTransFailed` case is only treated in the `default` case.

Could you test a patch like this ?
```
diff --git a/sw/airborne/peripherals/qmc5883l.c b/sw/airborne/peripherals/qmc5883l.c
index 8de97ddbb9..10ce42fb65 100644
--- a/sw/airborne/peripherals/qmc5883l.c
+++ b/sw/airborne/peripherals/qmc5883l.c
@@ -177,6 +177,9 @@ void qmc5883l_event(struct Qmc5883l *mag)
         /* End of measure reading, go back to idle */
         mag->status = QMC5883L_STATUS_IDLE;
       }
+      else if (mag->i2c_trans.status == I2CTransFailed) {
+        mag->status = QMC5883L_STATUS_IDLE;
+      }
       break;
 
     default:
```

alrochas1, 2025-10-01:
Thanks for the quick feedback.

You are right: the freeze happens because the driver logic never leaves `STATUS_MEAS` unless the transaction is `Success`. Yesterday we tested with my debug message + the watchdog function, and it actually triggered several times during a few-hour session (the `q5883l_errors_counter` was at 5 when we stopped). So the watchdog effectively recovered the driver each time.

We didn’t dig deeper yet, but this confirms that the state machine indeed gets stuck in `STATUS_MEAS` occasionally, and your proposed patch (handling `I2CTransFailed` explicitly) should fix it in a cleaner way.

We will test your patch on hardware the next occasion.

OpenUAS, 2025-10-16:
@alrochas1 any possibility to retest?. Else maybe we should ust integrate @gautierhattenberger workaround since no obvious negative side-effects. And a hanging mag is indeed the worst. We've seen that some batches of QMC5883L are not the most robust of hardware as you can see in the remarks in the rest of the code.

gautierhattenberger, 2025-10-16:
It would be better to test the one line patch on the driver before merging an additional watchdog that is maybe not necessary if the failure case is correctly processed.